bleskomat/bleskomat-diy #6: /bin/sh: 1: platformio: not found

ArbenKarb, 2020-07-25:
$ platformio --version
$ PlatformIO, version 4.3.4

Added path to bashrc

followed all steps

$ cd /bleskomat/bleskomat
$ sudo make install

cd ./device && platformio lib install
/bin/sh: 1: platformio: not found
make: *** [Makefile:52: install] Error 127

chill117, 2020-07-26:
Try without `sudo`:
```bash
make install
```

ArbenKarb, 2020-07-28:
Tried that, I get a permission error:

Error: Traceback (most recent call last):
  File "/home/admin/.local/lib/python3.8/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/admin/.local/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/admin/.local/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/admin/.local/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/admin/.local/lib/python3.8/site-packages/platformio/commands/lib.py", line 151, in lib_install
    lm = LibraryManager(storage_dir)
  File "/home/admin/.local/lib/python3.8/site-packages/platformio/managers/lib.py", line 40, in __init__
    super(LibraryManager, self).__init__(
  File "/home/admin/.local/lib/python3.8/site-packages/platformio/managers/package.py", line 604, in __init__
    os.makedirs(self.package_dir)
  File "/usr/local/lib/python3.8/os.py", line 211, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/usr/local/lib/python3.8/os.py", line 211, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "/usr/local/lib/python3.8/os.py", line 221, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/bleskomat/bleskomat/device/.pio'

chill117, 2020-07-29:
> File "/usr/local/lib/python3.8/os.py", line 221, in makedirs mkdir(name, mode)
> PermissionError: [Errno 13] Permission denied: '/bleskomat/bleskomat/device/.pio'

Looks like your current user lacks permission to create a new directory. Try this:
```bash
sudo chmod -R 0755 ./
```

closed 2020-08-01 by ArbenKarb