bleskomat/bleskomat-diy #13: Invalid API key signature

joetinker, 2020-10-12:
Please help me with resolving this error. HW "Beskomat" works, but after scanning QR wallet reports:

	Error message from remote provider: Invalid API key signature

Thanks

chill117, 2020-10-12:
Hello @joetinker. Where do you see that error? In the mobile wallet when attempting to scanning a QR code displayed on your TFT? Or do you see the error in your terminal window that is running the server?

joetinker, 2020-10-13:
Yes, in the mobile wallet (BLW or WoS) when attempting to scan a QR code displayed on Bleskomat. I generate key by "lnurl generateApiKey" on the server and in include id & key to the Bleskomat firmware.
The terminal window shows "lnurl:request GET /api?id=1bd03af8f5&n=1513154355&t=w&f=CZK&pn=1&px=1&pd=&s=645c0f7673ae25d93a8d09434ad96fa8a2e1a11f2acc1f16378aad53fbad53f6 +7m"

chill117, 2020-10-14:
> I generate key by "lnurl generateApiKey" on the server and in include id & key to the Bleskomat firmware

Did you put that API key in your server's config.json? It should be at `lnurl.auth.apiKeys` as an element of an array. For example:
```js
{
	"lnurl": {
		// ...
		"auth": {
			"apiKeys": [
				{
					"id": "rp/hlgc=",
					"key": "QA0ClaLFSjnsfybL8apu3YsDkvPAm8F9hq3bE8ZkNYw="
				}
			]
		}
	}
}
```
It might be necessary to use base64 encoded API key ID/Secret with the current device firmware because of how the device does the signing.

Try generating a new API key as follows:
```bash
lnurl generateApiKey --encoding "base64"
```
And add it to your server's config.

joetinker, 2020-10-14:
Thank you so much for the support, [ lnurl generateApiKey --encoding "base64" ] probably helped. I created a new id / key and transferred it on to another server using scp. Before that I transferred id / key via Windows clipboard, so maybe there was a change the encoding.

closed 2020-10-15 by joetinker