bleskomat/bleskomat-diy #19: Board is not countig accepted coins

martinkuchar2, 2021-04-27:
I have built my first Bleskomat, it is on and running (I can see 0 Kc on display) and the coin acceptor is programmed on and accepting coins (can see the right number on its display). But the device still shows 0 Kc event after inserting coins - looks like it can not accept signals from the acceptor. Please, can you help me to get this fixed? All grounds are connected in the right way (like on the diagram)... Thanks

chill117, 2021-04-27:
Have you seen these notes from the README file?
> There are two switches on the back of the HX-616 coin acceptor:
> * "NO" (normally open) / "NC" (normally closed) - This should be set to "NO"
> * "Fast" / "Medium" / "Slow" - This should be set to "Fast"

If these switches are not set as described here, then the pulses won't be sent in a way that the ESP32 can understand.

martinkuchar2, 2021-04-27:
Sure, this is set as it should be to NO and Fast. Any other idea, please?

chill117, 2021-04-27:
Do you have a shared ground between the ESP32 and the coin acceptor?

martinkuchar2, 2021-06-23:
> We've made some updates to the firmware and components list. The coin acceptor has been changed to the DG600F which is much more reliable than the previously used coin acceptor. Please refer to the latest project readme for details.

Please do you have a diagram of older version with 616 acceptor please? Can not find it here any more. Thanks

chill117, 2021-06-23:
You can find the old documentation here:
https://github.com/samotari/bleskomat-diy/tree/eccf6d46e87672ea28d633a8c26bc4cbd1c8c133

closed 2021-06-23 by chill117