bleskomat/bleskomat-diy #20: missing repo during make install

rmoh795, 2021-05-06:
Hi, 
i'm trying to compile the bleskomat, but when i try to do so, i get error:

remote: Repository not found.
fatal: repository 'https://github.com/chill1/lnurl/' not found

And this repo is indeed unavailable. Is there any workaround please?
Thank you

chill117, 2021-05-07:
Which command do you run when you see the error?

rmoh795, 2021-05-07:
It happens during Installing Libraries and Dependencies, when I run `make install` to install all dependencies and libraries. I am asked to enter my github login and then I get this error, that the repository is not found.

chill117, 2021-05-07:
Can you please copy/paste the output that you are seeing when you run the `make install` command?

rmoh795, 2021-05-07:
`└─$ make install 
cd ./device && platformio lib install
Library Storage: /home/user/Downloads/bleskomat-master/device/.pio/libdeps/bleskomat32
Looking for QRCode library in registry
Found: https://platformio.org/lib/show/1584/QRCode
LibraryManager: Installing id=1584 @ ^0.0.1
Using cache: /home/user/.platformio/.cache/e0/a4f1df25d468f09ebf2172c2033e85e0
QRCode @ 0.0.1 has been successfully installed!
Looking for TFT_eSPI library in registry
Found: https://platformio.org/lib/show/1559/TFT_eSPI
LibraryManager: Installing id=1559 @ ^2.2.6
Using cache: /home/user/.platformio/.cache/a2/12c42a822a24e2b336495dd8f584ada2
Unpacking  [####################################]  100%
TFT_eSPI @ 2.3.67 has been successfully installed!
LibraryManager: Installing lnurl @ 0.1.3
git version 2.30.2
Cloning into '/home/user/Downloads/bleskomat-master/device/.pio/libdeps/bleskomat32/_tmp_installing-jcchocr7-package'...
Username for 'https://github.com': rmoh795
Password for 'https://rmoh795@github.com': 
remote: Repository not found.
fatal: repository 'https://github.com/chill1/lnurl/' not found
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/chill1/lnurl', '/home/user/Downloads/bleskomat-master/device/.pio/libdeps/bleskomat32/_tmp_installing-jcchocr7-package']
make: *** [Makefile:58: install] Error 1``

chill117, 2021-05-07:
And which version of platformio are you running? Run the following to print the version:
```bash
platformio --version
```

chill117, 2021-05-07:
It looks like platformio is not even trying to pull the lnurl lib from the platformio registry - it looks to be going straight to GitHub. Try changing the line in [platformio.ini](
https://github.com/samotari/bleskomat/blob/master/device/platformio.ini#L62):
```
	chill1/lnurl@0.1.3
```
To:
```
	lnurl@0.1.3
```
Then try the install again. I'm guessing you're running an old version of platformio that doesn't recognize the namespace syntax.

chill117, 2021-05-07:
> But now, when I proceed with make compile I get error, that unknown package 'espressif32' is detected. Maybe it is caused by the version of platformio I have installed.

Yes, this is possible. If you are able to upgrade your platformio version, that might be the best solution.

rmoh795, 2021-05-07:
This issue has been resolved by removing the platformio installed by apt and by installation of the latest version of platformio (Instructions here: https://docs.platformio.org/en/latest/core/installation.html#installation-methods).

closed 2021-05-07 by rmoh795