bleskomat/bleskomat-diy #21: Support bills acceptor

iBobik, 2021-07-07:
Coins does not make sence for my use-case. I saw bills acceptor on some photos.

Does this project support it?
Are there some docs about it?

chill117, 2021-07-08:
Hello. You are correct in that this repo does not currently include support for a bill acceptor. We may add support for it during the second half of this year, but no guarantees. Until then you are welcome to fork and extend this project with a bill acceptor of your own - there are a few options available from online market places.

chill117, 2022-08-21:
@iBobik Support for the NV10 (USB+) and NV9 bill acceptors has been added by 6105a40e6dd995de38533d95790aa5229d335d9b and will be included in a future release.

chill117, 2024-04-16:
Closing because bill acceptor support was added.

closed 2024-04-16 by chill117