bleskomat/bleskomat-diy #24: DG600F serial communication. Not counting coins

relcead, 2021-11-10:
Hello. We build your project. It works really nice. Thank you for sharing it with us.
While building with your recent guide we noticed serial communication errors with DG600F. 

in the file: coin-acceptor.cpp
in the original line 77: Serial2.begin(COIN_ACCEPTOR_BAUDRATE, SERIAL_8N1, COIN_ACCEPTOR_SIGNAL, 0);

we had to change to:
line 77: Serial2.begin(COIN_ACCEPTOR_BAUDRATE, SERIAL_8E1, COIN_ACCEPTOR_SIGNAL, 0);

because serial communication in DG600F Manual:
S：Start bit 
B0 to B7：8 data bits 
C：Even check  
STOP：Stop bit 
It can set the transmit rate（1200bps/2400bps/4800bps/9600bps）in coin acceptor   
parameter A2. 

When using and testing different coin or bill acceptors the serial setup gets mixed up easy. It is expected behavour. It is not a bug or an issue. 

Use: SERIAL_8E1 with DG600F 

Thank you. Great DIY

chill117, 2021-11-11:
Thank you for bringing this to our attention. I will review the DG600F's manual and try my build with the suggested change.

relcead, 2021-11-11:
Thank you for your work.

chill117, 2022-03-25:
Tested and works w/ my hardware build. I've committed the change to the master branch. Thanks again!

closed 2022-03-25 by chill117